3DPrintDemon/Demon_Klipper_Essentials_Unified #85: When homing i get this "Error evaluating 'gcode_macro _HOME_Y:gcode'"

Trei-D, 2024-12-30:
I have updated through ssh the config from **2.9.1** to the current version, and followed the instruction per klicky_probe. But when tried to home, i have this error.

It used to work with previous configuration with no problems. 

Did I missed something during setup?

Setup:
- Printer: Biqu B1 running on a Raspberry Pi Zero 2W
- klipper v0.12.0-404-g80d185c9
- mainsail v2.13.2
- mainsail-config v1.2.1-1-gff3869a6
- moonraker v0.9.3-4-ga4604e33
- Demon_klipper_essential_unified previous version: 2.9.1

This is the error from the log:
```
Homing Y
Homing X
Homing Z
probe: open
toolhead: max_velocity: 9097.000000
max_accel: 1000.000000
minimum_cruise_ratio: 0.500000
square_corner_velocity: 5.000000
Error evaluating 'gcode_macro _HOME_Y:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _START_VARIABLES'
Traceback (most recent call last):
  File "/home/treid/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 13, in top-level template code
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 471, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _START_VARIABLES'
Error evaluating 'gcode_macro _HOME_Y:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _START_VARIABLES'

Error evaluating 'gcode_macro _HOME_Y:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _START_VARIABLES'
```
This is the code from klicky_macros.cfg
[image]

This is from homig_override section:
[image]

This is the result of the diagnostics from ssh:
[image]
[image]

3DPrintDemon, 2024-12-30:
Hi there, yes there has been changes to the systems for this release.

The error in your log shows you seem not to have the `demon_user_settings file` included in your system. It's looking for it but can't find it.

The klicky macro additions look ok 

The diagnostics say you're missing quite a bit there.

Firstly. Do you have all the files shown in this section the [documentation](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#the-file-structure-has-changed-for-v294-onwards)?

If the file structure doesn't look like in the link above you need to extract the files from the source directory inside the Other_Files directory so you have them there & then update your previous settings to the new files.

Depending on how you installed v2.9.4 you may need to add the updated variables file. If you used the new installer you should be ok.

If not please ssh this command....

```
cp -f ~/printer_data/config/Demon_Klipper_Essentials_Unified/Other_Files/Demon_Vars/demon_vars.cfg ~/
```

Also please make sure you have these includes in your printer.cfg
```
[include ./Demon_Klipper_Essentials_Unified/*.cfg]
[include ./Demon_User_Files/*.cfg]
```
There's also an extra on described in the documentation if you're wanting to use the Kiauh Shell extension.

You should also go through the documentation again to resolve the issue the diagnostic tool has highlighted for you.

Trei-D, 2024-12-30:
It was my bad, as you said I haven't fully went through documentation,I sorted it out.

Now I'm running in another problem, when I start printing, just before making the adaptive mesh, this error pops up:
```
Unknown command:"_HEAT_WAIT"
Error evaluating 'gcode_macro _ADAPTIVE_LEVELLING:gcode': gcode.CommandError: This error is caused by your printer not supporting this feature! Please enable Manual Levelling in the demon_user_settings.cfg file
Traceback (most recent call last):
  File "/home/treid/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/treid/klippy-env/lib/python3.9/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 42, in top-level template code
  File "/home/treid/klipper/klippy/extras/gcode_macro.py", line 89, in _action_raise_error
    raise self.printer.command_error(msg)
gcode.CommandError: This error is caused by your printer not supporting this feature! Please enable Manual Levelling in the demon_user_settings.cfg file
Error evaluating 'gcode_macro _ADAPTIVE_LEVELLING:gcode': gcode.CommandError: This error is caused by your printer not supporting this feature! Please enable Manual Levelling in the demon_user_settings.cfg file

Error evaluating 'gcode_macro _ADAPTIVE_LEVELLING:gcode': gcode.CommandError: This error is caused by your printer not supporting this feature! Please enable Manual Levelling in the demon_user_settings.cfg file
```
I have checked the file, adaptive meshing is enabled, the requirements for klipper adaptive leveling are in place.

3DPrintDemon, 2024-12-30:
Hi thats ok, glad you sorted that issue out.

The next one is because you haven't included the heat soak file for your printer in your printer.cfg file. The Diagnostic tool will tell you what needs doing, then you should follow its advice.
You can also check the prerequisites section of the documentation to see if you have any missing files.

Trei-D, 2024-12-30:
I saw that, but my printer came from factory with neopixel led on the toolhead. I have managed to make it wor in the 2.9.1 version by adapting the leds cfg. I have included the heat soak file as well.
[image]
This is the cfg responsible for led:
```
[neopixel sb_leds]
pin: P1.24
#   The pin connected to the neopixel. This parameter must be provided.
chain_count: 2
#   The number of Neopixel chips that are "daisy chained" to the
#   provided pin. The default is 1 (which indicates only a single
#   Neopixel is connected to the pin).
color_order: GRB
#   Set the pixel order required by the LED hardware. Options are GRB,
#   RGB, GRBW, or RGBW. The default is GRB.
initial_RED: 1.0
initial_GREEN: 0.0
initial_BLUE: 1.0
initial_WHITE: 0.0
#   Sets the initial LED color of the Neopixel. Each value should be
#   between 0.0 and 1.0. The WHITE option is only available on RGBW
#   LEDs. The default for each color is 0.#

# Most configuration for the macros can be done by modifying the variables in the _sb_vars macro
# at the start of this file.

##########
# MACROS #
##########

# The following status macros are available (these go inside of your macros):
#
#    STATUS_READY
#    STATUS_OFF
#    STATUS_BUSY
#    STATUS_HEATING
#    STATUS_LEVELING
#    STATUS_HOMING
#    STATUS_CLEANING
#    STATUS_MESHING
#    STATUS_CALIBRATING_Z
#
# With additional macros for basic control:
#
#    SET_NOZZLE_LEDS_ON
#    SET_LOGO_LEDS_OFF
#    SET_NOZZLE_LEDS_OFF
#
# Contributed by Voron discord users wile.e, Tetsunosuke, and etherwalker

[gcode_macro _sb_vars]
# User settings for the StealthBurner status leds. You can change the status colors and led
# configurations for the logo and nozzle here.
variable_colors: {
        'nozzle': { # Colors for nozzle states
            'busy': {'r': 0.0, 'g': 0.0, 'b': 1.0, 'w': 0.0},
            'cleaning': {'r': 0.0, 'g': 0.02, 'b': 0.5, 'w': 0.0},
            'calibrating_z': {'r': 0.8, 'g': 0., 'b': 0.35, 'w': 0.0},
            'heating': {'r': 0.3, 'g': 0.18, 'b': 0.0, 'w': 0.0},
            'homing': {'r': 0.0, 'g': 0.6, 'b': 0.2, 'w': 0.0},
            'leveling': {'r': 0.5, 'g': 0.1, 'b': 0.4, 'w': 0.0},
            'meshing': {'r': 0.2, 'g': 1.0, 'b': 0.0, 'w': 0.0},
            'off': {'r': 0.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'printing': {'r': 1.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'standby': {'r': 0.0, 'g': 1.0, 'b': 0.0, 'w': 0.},
            'on': {'r': 0.8, 'g': 0.8, 'b': 0.8, 'w':1.0},
        },
        'thermal': {
            'hot': {'r': 1.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'cold': {'r': 0.3, 'g': 0.0, 'b': 0.3, 'w': 0.0}
        }
[... 75 lines cut ...]
    _set_sb_leds_by_name leds="nozzle" color="leveling" transmit=1

[gcode_macro status_homing]
gcode:
    _set_sb_leds_by_name leds="nozzle" color="homing" transmit=1

[gcode_macro status_cleaning]
gcode:
    _set_sb_leds_by_name leds="nozzle" color="cleaning" transmit=1

[gcode_macro status_meshing]
gcode:
    _set_sb_leds_by_name leds="nozzle" color="meshing" transmit=1

[gcode_macro status_calibrating_z]
gcode:
    _set_sb_leds_by_name leds="nozzle" color="calibrating_z" transmit=1

[gcode_macro status_printing]
gcode:
    _set_sb_leds_by_name leds="nozzle" color="printing" transmit=1
```
and i'm using SB_LEDs_heat_soak.cfg for the heat soak. 

Is there something that i have missed again from the docummentation?

3DPrintDemon, 2024-12-30:
Oh I just saw also you had two messages previously there!
It seems your printer doesn't have Z tilt adjust or QGL functions. If that's the case please enable this line in your `demon_user_settings file`.

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/8d45fa4d8325bf795b0086f66af39b09dda9090e/Other_Files/DEMON_User_Files_SOURCE/demon_user_settings_v2.9.4.cfg#L25

If your LED's work then thats fine, the diagnostic tool looks for the default files suggested for use with the macros with default names. It says at the top that if names are changed (or indeed if different files are used) the this will cause false reports of missing files.

Trei-D, 2024-12-31:
Everything works now, just something curious is happening. It take few minutes to start the print, I think is the heat soak timer or something related to that.
I want to thank you for your good work and the impressive amount of time invested in this time saver klipper addon.

3DPrintDemon, 2024-12-31:
Hi great it’s all working now. You’re most welcome, this project has been ongoing for the last year or so & has grown is scale & popularity, hopefully it’ll continue to do so. 
Yes the macros will run a heat soak timer to allow the printer to fully warm up & heat soak the bed & probe. 

You can toggle it off/on with the macro button in mainsail. It’ll stay that way until you restart Klipper. There’s heat soak timers in the demon_user_settings file you can adjust. Also there’s an option called “start_my_print_already” that’ll disable the timers until you toggle them on again (until restart) or change that option in the file again.

closed 2025-01-05 by 3DPrintDemon