3DPrintDemon/Demon_Klipper_Essentials_Unified #86: All files in ~/printer_data/config/Demon_Klipper_Essentials_Unified/... are read-only

bmberetta, 2024-12-30:
I upgraded from v2.9.3 to v2.9.4 via the ```Update Manager``` on Mainsail's ```machine``` page. I now see all the files in the ```~/printer_data/config/Demon_Klipper_Esentials_Unified``` directory as read-only in the Mainsail interface. The permissions on the individual files are set at mode 644 and are correctly owned, and in the correct group. Files outside of this directory are editable. I can edit the files by SSH'ing to the RPi. Is this by design, or has something gone awry?

3DPrintDemon, 2024-12-30:
This is due to the use of the mainsail update manger & is expected behaviour if this system is used. If you edit files you’ll get a “dirty” notice on the update manager section & files will be cleaned on the next update. You can either use the old method & manually update & not use the update manager, or copy files out & mess around with the includes etc if you wanted. It’s not an issue with the files or your printer, just the method of updating them in mainsail.

bmberetta, 2024-12-31:
So a "feature", not a bug! I understand. Thanks for the quick response.

3DPrintDemon, 2024-12-31:
You’re welcome, yes it’s the way the update manager maintains the directories under its control, it’s the system’s normal behaviour. 😎

closed 2024-12-31 by 3DPrintDemon