3DPrintDemon/Demon_Klipper_Essentials_Unified #88: Can not pass Demon Diagnostic checks

lunarconcepts, 2025-01-02:
New to submitting issues but didn't see any other source for answers? 

New install of DKEU on Dec. 24. Overall I was up and printing and I am just trying to get some housecleaning done. I think its working mostly but I didn't know if there was a Functions Checklist just to verify that everything is as expected. I ran the Demon Diagnostic today and I have  a handful of issues I cannot resolve.  Under the Macro System Check I have  errors for multiple  .cfg files exist but are not included. They are included in my printer.cfg so do they need to be somewhere else? Example is my KAMP_LiTE.cfg. It is erroring but my smart park and purge are working as expected? 

In my Printer.cfg Checks I have a few items that are spelled different in the check than what Klipper with allow.  I am getting an error for  undefined exlude_objectS when I have [exclude_object]. It is also flagging saved_variables and force_move as not defined but I have them in my p.cfg as instructed?  Screenshots below.  This was my biggest digital upgrade to my Voron and its likely I am just making mistakes. Thanks for any help. 

[image]
[image]
[image]

3DPrintDemon, 2025-01-02:
Hi the search terms for the diagnostic tool are strict, it reads the file & not what Klipper sees at the moment. So anything on the same line as the include will been seen as a deviation from what should be there. 

I’d recommend putting your comments for your includes on the line above, or grouping your includes together with a single comment on a separate line top & bottom. This will give you the green results you’d like to see.

lunarconcepts, 2025-01-02:
Thank you! Like so much with learning, so much time spent when the fix was something so simple. I was trying to keep my self organized and ended up commenting myself in the foot.

lunarconcepts, 2025-01-02:
I removed all commenting, save/restart and had to power cycle and  still have the same errors on the same includes and definitions? I'm still doing something wrong?
[image]
[image]
[image]

3DPrintDemon, 2025-01-02:
Hmm don’t see anything wrong. Any white space (spaces) on the end of the include lines, maybe it’s picking up on that?
Try removing them & see.

3DPrintDemon, 2025-01-02:
Hi great that got it! Yeah the diagnostics are very strict, they only like very specific lines there. To Klipper it doesn’t matter if you have a hashed comment after the defined line or white space as it’s not looking that closely & will function ok. I tried it like this as sometimes someone might have something else on the line of the include or system function. This way it’s only happy if it’s exact. I could look at easing that but it then possibly leaves things open for the diagnostics missing stuff. 
I’m glad the system is working & thank you for your support, it’s very much appreciated.