3DPrintDemon/Demon_Klipper_Essentials_Unified #90: Using an assumed user name in path

bmberetta, 2025-01-03:
I updated to v2.9.4-139 today. At the utilities startup I am getting the following error `5:59 PM
/usr/bin/bash: /home/pi/printer_data/config/Demon_Klipper_Essentials_Unified/Other_Files/Demon_User_Files_Updater/Demon_Disk_Space_Checker.sh: No such file or directory`

I don't use the user "pi". 

Should not that call rather be `~/printer_data/config/Demon_Klipper_Essentials_Unified/Other_Files/Demon_User_Files_Updater/Demon_Disk_Space_Checker.sh` to prevent such issues? i.e. use `~` instead of `/home/pi`.

3DPrintDemon, 2025-01-04:
You’d think wouldn’t you, that was indeed my original intent to do it as you suggest. Unfortunately the Kiauh shell script extension requires a full path starting as it does currently with a full path or it won’t work. Hence the multiple files with different stock user names for 90% of people to use. 
So sadly if you want to use the shell system you’ll have copy the file out to the new Demon_User_Files dir & include it there instead so you can edit it to your file path.

bmberetta, 2025-01-04:
That's seems a bit a large assumption on the part of the KIAUH developer. Seems the best recourse for me is to switch to using "pi" as the username on my klipper system. I'd rather not have a "spaghetti" situation that I have to keep track of every time I upgrade.

Thanks for the quick reply and suggestion on how to remedy the situation.

3DPrintDemon, 2025-01-04:
Yes it’s rather unfortunate it needs to be the full path have to say, it would’ve been much easier to use ~/, it does make it very difficult for people who use nonstandard user names. 

You’re welcome.

bmberetta, 2025-01-04:
Thanks again for the excellent utilities you've created! I'll be changing usernames as I want to keep using these utilities in the future.

closed 2025-01-04 by bmberetta

3DPrintDemon, 2025-01-04:
I’m glad you like my work & wish to continue using the macros. Happy printing!