3DPrintDemon/Demon_Klipper_Essentials_Unified #92: IDEX?

kdm303, 2025-01-04:
Hi,

First let me say you've put a great package together here. I didn't think anything could convince me to change from my custom configs.

It took a couple days of playing but now I've got everything worked out on my CoreXY and overall it's working pretty great. Just a couple minor issues I think I can work out.

I've started setting up my Ender 5+ which shouldn't take long and I've got a Klipper enabled Tenlog TL-D3-Pro with Eddy I've been hacking on and I'd love figure out how to adapt it.

Any plans to support IDEX or things I should beware of?

Thanks!

3DPrintDemon, 2025-01-04:
Hi thanks, glad you're enjoying my work here. Sounds like you're having fun converting your printers over!

I'm not too sure about any IDEX support as I don't have any suitable machines to test anything like that on unfortunately. I'm hoping to add some multicolour functionality soon maybe if things go to plan, nothing confirmed as yet but it's something I'd love to incorporate.

kdm303, 2025-01-04:
Just started printing my last plate of parts for a Box Turtle MMU and will have the MCU for it worked out soon.

There aren't a tremendous number of differences between multiple head and filament configurations. I'd be happy to help testing with either of those things if/when you get to that point.

3DPrintDemon, 2025-01-05:
Thats fantastic I've seen a few videos of the box turtle, it looks very good.

Yes I'm sure the integration of a multicoloured setup shouldn't be too difficult. Once I get onto this I let you know.

closed 2025-01-12 by 3DPrintDemon