3DPrintDemon/Demon_Klipper_Essentials_Unified #93: Misc issues

kdm303, 2025-01-04:
Couldn't find reference to what encoder sensor should be named? currently "encoder_sensor"

Irrelevant typo but OCD tingling: demon_user_settings_v2.9.4.cfg line 479 "BTT Smart _Senor_ True/False"

variable_purge_along_y set to true but purges on X, testing the inverse after I get an Eddy calibration on heated bed.

Klippy log contains a lot of "Neopixel update did not succeed" entries, any ideas?

kdm303, 2025-01-04:
Purges on X regardless of variable_purge_along_y setting.

3DPrintDemon, 2025-01-05:
1. Please see [this section.](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#filament-sensor)
2. Typo fixed, many thanks
3. Legacy Static Purge along Y works perfectly as intended. Please see image below, I would recommend double checking your setup & looking for any other enabled purge macros in your system.
4. Neopixel issues sounds like your neopixels are not named or numbered or configured correctly, please double check that system also.

This printer is using the exact same files downloaded & updated via Mainsail Update Manager as you should have on your printer. Two prints started back to back with only the variable_purge_along_y changed 

[image]

closed 2025-01-12 by 3DPrintDemon