3DPrintDemon/Demon_Klipper_Essentials_Unified #94: Pause/Resume

kdm303, 2025-01-05:
Unable to resume after pause:

Console log shows error:

Error evaluating 'gcode_macro RESUME:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'filament_sensor'

Using BTT SFS2.0:

[filament_switch_sensor filament_sensor]
switch_pin: ^PC15
pause_on_runout: False
runout_gcode:
  PAUSE
  M117 Filament Runout Detected
insert_gcode:
  M117 Filament Insert Detected
event_delay: 3.0
pause_delay: 0.5

[filament_motion_sensor encoder_sensor]
switch_pin: ^PF1
detection_length: 10
extruder: extruder
pause_on_runout: False
runout_gcode:
  PAUSE
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted

3DPrintDemon, 2025-01-05:
That is the Mainsail RESUME macro & not a Demon macro calling that error as it is performing a check to ensure filament is still loaded as you instructed it. Please double check your setup & system.

kdm303, 2025-01-05:
Figured it out. Defining the sensors in demon_user_settings.cfg is all that's needed.

variable_runout_sensor: True and variable_encoder_runout_sensor: True

If it's also defined in My_Macros.cfg's _CLIENT_VARIABLE bad things happen.

closed 2025-01-05 by kdm303

3DPrintDemon, 2025-01-05:
I’m curious. As I mentioned the mainsail RESUME macro is not a demon macro so it requires no reference or settings from the demon_user_settings file to be on or off, it looks directly at the filament sensor that’s defined within your printer.cfg file itself & is totally independent in that regard. 

The demon_user_settings file variables for 

variable_runout_sensor: variable_encoder_runout_sensor:

only tell the demon macros to perform their own checks & verifications against the filament sensor’s state at different times. Those options have no say or function if your sensor is defined with the Mainsail _CLIENT_VARIABLE macro during the operation of the Mainsail RESUME macro. 

If you remove your filament sensor name from the  Mainsail _CLIENT_VARIABLE macro all it does is not check if filament is still loaded before letting Klipper RESUME the print.