3DPrintDemon/Demon_Klipper_Essentials_Unified #95: SV08 instructions.  [save_variables]

gregma, 2025-01-05:
In the SV08 section, you state to comment out the 
[save_variables]
filename = ~/printer_data/config/saved_variables.cfg

and replace with:
[save_variables]
filename: ~/demon_vars.cfg

Do you really mean to put that in the home directory and not in printer_data/config?

3DPrintDemon, 2025-01-05:
Yes as it not a file the user needs to see or interact with. However, it’s your printer you can place it where you please.

gregma, 2025-01-05:
> Yes as it not a file the user needs to see or interact with. However, it’s your printer you can place it where you please.

Nope, if it's good there, then I'm good having it there :)

closed 2025-01-05 by gregma

3DPrintDemon, 2025-01-05:
Cool no worries. The installers will deal with any installation or updates for it. It’s also easily modified/copied/moved or deleted by using the ssh terminal or by visually browsing the file structure with any good SFTP client software.