3DPrintDemon/Demon_Klipper_Essentials_Unified #99: Neopixel Caselight

Graf911, 2025-01-09:
I really liked your shell
I used a klippain shell before that.
there was an opportunity to choose the lighting option of the printer with effects neopixel
right now I can't get my neopixel to work, two strips of 18 LEDs in parallel.
The LEDs on the print head are working properly
maybe you have a file or an instruction for the dumb

3DPrintDemon, 2025-01-09:
Hi glad you like the shell, the led neopixel file is the Voron Stealthburner control system. In this system you need to name the LEDs correctly (as I’m sure you’ve already done) & then add the sting count to the file, then you need to allocate which LEDs function as the nozzle or logo leds as it’s originally meant for Toolhead LEDs. Works fine for all others too just have to work around the naming. 

Chain count here:
https://github.com/3DPrintDemon/Voron-Stealthburner/blob/a364aaf69e510c7cb60a97036ede3d7f476abb8f/Firmware/stealthburner_leds.cfg#L58

LEDs used for logo functions:
https://github.com/3DPrintDemon/Voron-Stealthburner/blob/a364aaf69e510c7cb60a97036ede3d7f476abb8f/Firmware/stealthburner_leds.cfg#L130

LEDs used for nozzle functions:
https://github.com/3DPrintDemon/Voron-Stealthburner/blob/a364aaf69e510c7cb60a97036ede3d7f476abb8f/Firmware/stealthburner_leds.cfg#L135

3DPrintDemon, 2025-01-09:
However… 

if you’re running 2 chains, one for the toolhead & another for top hat or skirt then you’ll need this file. 

https://github.com/3DPrintDemon/Voron-Stealthburner/blob/main/Firmware/Neopixel_Control.cfg

Then you need to set the values as above here for both chains. 

Toolhead pin line 56
Toolhead chain count line 58
Top hat pin line 77
Top hat chain count line 78

Then set the led allocations from line 150

igeighty, 2025-01-13:
thanks, giving it a shot as led_effects giving timer errors. i had two extra strips also, this will be handy. and not so demanding as led effects