3DPrintDemon/Demon_Klipper_Essentials_Unified #100: UNLOAD_FILAMENT:  'core_vars' is undefined

mverrilli, 2025-01-11:
**Describe the bug**
While issuing UNLOAD_FILAMENT, received an error: 

```Error evaluating 'gcode_macro UNLOAD_FILAMENT:gcode': jinja2.exceptions.UndefinedError: 'core_vars' is undefined```

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh start
2. UNLOAD_FILAMENT
3. See previously mentioned console error

**Expected behavior**
No error, filament unloads. 

**Printer (please complete the following information):**
 - Printer brand: Sovol
 - Printer model: SV08
 - Mainline 

**Additional context**

Fixed the issue by adding to UNLOAD_FILAMENT: 

```  {% set core_vars = printer["gcode_macro _CORE_VARS"] %}```

3DPrintDemon, 2025-01-11:
Ahh nicely spotted there. That one slipped by, if you hit update it should be sorted now.

mverrilli, 2025-01-11:
Thanks, will check it out later as I have a long print running right now. 

Great job on this project, by the way. Very well organized, clear instruction, and really good features.

3DPrintDemon, 2025-01-12:
Thanks so much I'm glad you like my work here.

closed 2025-01-12 by mverrilli