3DPrintDemon/Demon_Klipper_Essentials_Unified #101: SV08 Mainline Screen mod failure

BrunoSonego, 2025-01-11:
**Describe the bug**

not compiling after loading the SV08_ML_display_V1.0.1.cfg
error was invalid input on "[menu __main __control2 __ledonoff]"

commenting the lines made it able to run and start

#[menu __main __control2 __ledonoff]
#type: disabled

**Printer (please complete the following information):**
 - Printer brand Sovol
 - Printer model SV08
 - Mainline Klipper

3DPrintDemon, 2025-01-12:
Hi, thanks for posting, are you able to tell me the defined name for your white LED pin in your printer.cfg please.

BrunoSonego, 2025-01-12:
I tried with both config original mainline klipper and the one used before on stock sovol, ended up using the same as used in the macro config

# [led main_led]
# white_pin:PA3
# cycle_time: 0.010
# hardware_pwm: False
# initial_WHITE: 1.0

[output_pin Printer_Lights]
pin:PA3
pwm: 1
value:1
cycle_time: 0.01
shutdown_value: 0

3DPrintDemon, 2025-01-12:
Ok great thanks for the confirmation & letting me know. I can edit the file & upload a new version tomorrow.

BrunoSonego, 2025-01-12:
No need to rush its just disabling the LED ON/OFF button on the HMI for now. i can test it once you edit. Thanks for keeping it up

3DPrintDemon, 2025-01-12:
The light control should be called Printer_Lights & should be in the pins section & not listed as a led button. 

There should be a Printer_Lights option in the small stock screen menu system

3DPrintDemon, 2025-01-12:
New file with lines commented out uploaded

closed 2025-01-12 by 3DPrintDemon