3DPrintDemon/Demon_Klipper_Essentials_Unified #102: Custom Variables

mmrrxx00, 2025-01-12:
This might already be in the macros but I can't seem to find it. I am setting up a box turtle on  my system and I need to add 

{% set initial_tool = params.TOOL|int %} to the start macro. I tried adding it using the custom expansion but it did not seem to add it.  Later in the start macro when I tried to call T{initial_tool} it just gave an error. I can call the T0 or whatever number fine form the console but it does not seem to work in the start print macro. I tried to test to see the variable was not being set but the file is read only so I could not edit it for testing.

3DPrintDemon, 2025-01-12:
Hi. Thanks for posting. Did you add the line below to your slicer’s start gcode, make sure it’s on the end of the last line with the other parameters, not as a new line. 

```
TOOL={initial_tool}
```

mmrrxx00, 2025-01-12:
Sorry yes I forgot to mention that was added. It works fine if I hard code in the T0 command as a custom pre print. I also checked the gcode file and orca is adding the tool in there. The T0 works for now since I am just getting it setup but was trying to avoid being limited to having to always start with slot 1 if possible.

3DPrintDemon, 2025-01-12:
If you'd like to test your own mods you can disable the update manager for the macros temporarily & then download the demon_print_start_end cfg file from your printer to your pc.
 Make your changes to it then drop it back into the demon macro directory. This will replace the stock file & allow you to test changes.

Then when you're done you can simply re-enable update manager for the macros, it'll tell you their status is "dirty" & you can "soft Recover" your current version back to stock.

mmrrxx00, 2025-01-12:
I got it figured out. It works fine if you put the variable and the macro call all in the same section. I had been trying by setting the variable in the preprint then calling the command later on in the start macro. Sorry to bother you with this. Keep up the great work these macros make life so much easier.