3DPrintDemon/Demon_Klipper_Essentials_Unified #103: Existing config autosave is corrupted. Can't complete SAVE_CONFIG

BrunoSonego, 2025-01-12:
**Describe the bug**
Existing config autosave is corrupted. Can't complete SAVE_CONFIG

**To Reproduce**
using command SAVE_CONFIG

**Screenshots**
```
#[save_variables]
#filename = ~/printer_data/config/saved_variables.cfg #Original

[save_variables]
filename: ~/demon_vars.cfg
```

**Printer (please complete the following information):**
 - Printer brand Sovol
 - Printer model SV08
 - Mainline Klipper

BrunoSonego, 2025-01-12:
Issue was that demon_vars.cfg has not been created in the main folder during install

BrunoSonego, 2025-01-12:
*update. actually I'm still with the issue after manually creating the demon vars file

BrunoSonego, 2025-01-12:
[image]

3DPrintDemon, 2025-01-12:
The command save_config has nothing to do with your save_variables file. 

The save_config command is to do with Klipper directly not the macros. 

Also the demon save variables file is not created in the config directory, it is created & used in the user’s home directory.

It’s telling you you have edited or changed the auto save area of your printer cfg. 

Please check you have not modified this area or you will get this error. It is not an issue with the macros but an issue with your printer.cfg file.

3DPrintDemon, 2025-01-12:
The printer.cfg auto save area is this area at the very bottom of your printer.cfg

https://github.com/Sovol3d/SV08/blob/a60644875f8c756d20b3828c9416518b414b5491/home/sovol/printer_data/config/printer.cfg#L446

if you have any spaces or manual changes made here that are not correct will result in your above error. 

Again this is not an issue with the macros but a change or modification you made to your printer.cfg file.

closed 2025-01-12 by 3DPrintDemon

BrunoSonego, 2025-01-12:
Thank you for the help. it really was this. I got some of the added parts after the save area and it was blocking the write action.